TheAlbertDev/MASB #2: NODE_F401RE not found. Please ensure the device is correctly connected.

Laiaborrell, 2021-02-11:
Hola Albert,
tinc la placa connectada i de fet (vull suposar) que funcionant perquè emet llum. Tot i així, em surt el següent error (que adjunto a continuació) cada cop que pujo el programa.
[image]
[image]

He buscat l'error a internet però no ho he pogut solucionar. 
Mersi,
Laia.

Albert-Alvarez, 2021-02-13:
When you connect the board, does a window open as if you had connected a pen drive? When you try to upload the program, does the LED COM blink?

It is difficult to give you solutions without more information 😅 I give you some tests that you can do before we look at it in the next session if you can't figure it out:

- Try using another USB port.
- Use another USB cable (there are cables that are only for charging and do not include the data bus).
- Make sure you pick the appropriate upload method: Upload method > Mass Storage.
- Reinstall/update the board drivers in the Arduino IDE.
- Reinstall Arduino IDE.

Laiaborrell, 2021-02-13:
No, it does not appear as a pen drive was connected. Maybe the USB cable is just for charging. I do not have more cables but as soon as I get another one I will try again and inform you.
Thankyou

Albert-Alvarez, 2021-02-13:
> No, it does not appear as a pen drive was connected. Maybe the USB cable is just for charging. I've tried with another cable and it works.
> 
> Thankyou

Glad to hear that the proposed solutions have helped you to solve the problem 😉

closed 2021-02-13 by Albert-Alvarez