TheAlbertDev/MASB #4: nothing to commit, working tree clean

hrodrigo7, 2021-02-17:
Hello Albert,
I'm trying to do the stm of the first practice but when I change the code and try to do a 'git commit' it tells me that there is nothing to commit (I attach a photo). I have not executed the git commit command before for this new version.
[image]

Albert-Alvarez, 2021-02-17:
If git tells you that there is nothing to commit that's because there isn't any file in the stage. The files are added to the stage using the git add command, which you have already done. So, that means that from the last commit you have not edited any file in the repository, no files have been added to the stage area with the git add command, and, therefore, there is nothing to commit.

Check that there are any files in your repository folder (maybe you have created the project in another folder accidentally), or that you have already edited any file.

Attach to this issue an image of the terminal for the output of the git log and git status, respectively.

hrodrigo7, 2021-02-18:
The fact is that I did have edited the files in the repository and there should be something to commit. I have also thought that it could be that the project is in another folder but I think it is in the correct place (I also attach the folder of the project).

The images of the errors and the folder are attached in the main issue.

Albert-Alvarez, 2021-02-18:
It is difficult to give you a solution without being able to have the computer in front. If you agree, I'll send you an email to meet virtually this afternoon, we'll see it and then I'll give the answer here so that the rest of my colleagues also have the answer.

Meanwhile, install [TeamViewer](https://www.teamviewer.com/es/) and Microsoft Teams (available as student on [https://nuvol.ub.edu/](https://nuvol.ub.edu/)) on your computer.

Albert-Alvarez, 2021-02-18:
In the remote assistance, the above-mentioned facts were not observed and everything seems to be working correctly. I close this issue and, if you have any further questions, do not hesitate to open a new one.

closed 2021-02-18 by Albert-Alvarez