TheAlbertDev/MASB #6: Problemes al fer commit

juliameca, 2021-02-21:
Hola Albert, 

Estic intentant penjar els REPORTS però em dona el següent error:

[image]

Com diu la descripció de l'error, vaig esborrar l'arxiu i el vaig tornar a copiar. Però em seguia sortint el mateix error. Llavors he vist que no estava a la branca A-julia, els números que m'apareixen en color blau són l'identificador del RETO STM, tal com pots veure a la següent imatge: 

[image]

He intentat accedir-hi amb la comanda checkout -b develop/A-julia (amb la que vaig crear la branca) però tampoc em serveix de res perquè clar, em diu que ja està creada. De fet, quan li poso "git status" em diu que "HEAD is detached at 03ba27e" (que és el reto). 

[image]

Em podries dir com posar-lo a la branca correcta? 

Moltes gràcies i disculpa les molèsties, 

Júlia Meca.

Albert-Alvarez, 2021-02-21:
Issues' title and content in English, please.

juliameca, 2021-02-21:
Okay, sorry. It's translated now.

Albert-Alvarez, 2021-02-21:
Hi Julia,

To switch to other branch you have to execute the command:

```bash
git checkout <branch-name>
```

In this case, without the `-b` parameter. This parameter is indicated to create a branch and, then, switch to the created branch. As your branch already exists, you must not add the `-b` parameter.

Let me know if you have been able to switch to your branch and if your problems have been solved by doing so.

juliameca, 2021-02-21:
Hello again, 

I tried the command you mentioned. I didn't know for sure if I only had to write "A-julia" or "develop/A-julia" as the name of the branch. So I tried both. 

In both cases, I get the same error as before. Look: 

[image]

What should I do? Thank you.

Albert-Alvarez, 2021-02-21:
It seems that, as the information in the terminal indicates, a git process has crashed. Git creates a lock file when it starts a process and deletes it when it finishes. It does this to prevent other git processes from running if one is already running (by checking whether such a lock file exists before starting a process). Seems that one of those processes has crashed before finishing and the lock file has been left without deleting it. Let's remove it. Execute the following command and let us know if this has solved your problem.

```bash
rm -f .git/index.lock
```

juliameca, 2021-02-21:
After trying this command, It still does not recognize the branch A-julia. Moreover, I tried to access the arduino folder I have inside masb-p01-blaugrana; but it says it doesn't exist. 

[image]

To ensure the REPORT files were still there, I ran a "git status" and confirmed they are still there (in fact, it even recognized the folder arduino here)

[image]

Albert-Alvarez, 2021-02-21:
But your branch is `A-julia` or `develop/A-julia`?

juliameca, 2021-02-21:
When I created the branch, I wrote: 

`git checkout -b develop/A-julia`

So I believe my branch is called A-julia. Nevertheless, I tried both options and none of them worked. It returns the same error: "A-julia did not match any file(s) known to git". But as you can see in the next image, the branch is created. 

[image]

Albert-Alvarez, 2021-02-21:
Your branch is called `develop/A-julia`, so you have to use the command:

```bash
git checkout develop/A-julia
```

to jump to your branch.

juliameca, 2021-02-21:
Okay, I do so. 

[image]

Even then, a new error pops up. How can I fix this new one?

Sorry for the inconvenience.

Albert-Alvarez, 2021-02-21:
You don't have to apologize! 😉

Git is telling you that you have edited a file in your current commit (03ba27e). In this case, the file `masb-p01.ino`. So, it doesn't allow you to switch to another branch without saving this modifications.

If you want to hold the modifications, the easiest way to solve this (not the most correct...) is:

1. Copy/paste the file `masb-p01.ino` to other location outside the folder of your repository (outside the `masb-p01-blaugrana` folder).

2. Restore your file to its original state with this command:

```bash
git checkout -- arduino/masb-p01/masb-p01.ino
``` 

3. Switch to your branch:

```bash
git checkout develop/A-julia
```

4. Copy/paste the previously copy/pasted file from outside your repository to the folder `arduino/masb-p01/`.

juliameca, 2021-02-21:
Thank you! The error doesn't appear anymore! :)

closed 2021-02-21 by Albert-Alvarez