TheAlbertDev/MASB #7: LED arduino help

mariamor99, 2021-02-21:
Hi Albert! 
I've been trying to see why the code doesn't work properly but I can't find the error. 
When I press the button, the LED starts blinking and if I press it again, it turns off but if I press it once more to start blinking again, it does nothing. It's as if the loop only worked once. 
I also tried putting a _condition = true_ at the end of the code but then it doesn't work at all. 
I attach an image of the code to see if you can guide me a little.
Thank you so much!

[image]

Albert-Alvarez, 2021-02-21:
Hi Maria,

First of all, try to add code in your issues as code. I mean, not using images. In this [page](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks) you can see how to do it. Simply, copy/paste your code inside a code block (you can also click in the <kbd>Insert code</kbd> button in the issue editor).

I say this because one thing that I wanted to do was to check if you haver your curly brackets `{` `}` correctly paired, and that's something that I can't to with an image 😅

Besides that, I've seen in your image that you set to false the variable condition (`condition = false`) when you push the button for a second time (to switch off the LED). Doing so, when you push again the button, the condition in the while loop `condition == true` doesn't meet and, then, the LED will not blink. Check this.

Finally, even if you solve this, in the second time that you read the state of the button, you don't check if you've already pushed it previously. You are performing a level-triggered event (not edge-triggered). Check this too.

Don't worry if you are not able to achieve the challenge. I will review it in your Pull Request and provide you more details about how to solve it. The important thing is that you upload your try in other to have a basis on which to base my corrections.

mariamor99, 2021-02-21:
Ok! I'll try to solve it!
Thank you

closed 2021-03-06 by github-actions[bot]