TheAlbertDev/MASB #8: Commit Reports on different branches

juliameca, 2021-02-21:
Hello again Albert. 

I'm trying tu upload the REPORTS in the arduino and stm32cube folders. 

In the develop/A branch, I `add` and `commit` the arduino's REPORT and the terminal "frozes" (I cannot write and I'm forced to close the terminal) with this message: 

[image]

On the other hand, when I `add` and `commit` the stm's REPORT in the develop/B-branch. An analogous message appears too: 

[image]

No report appears in the GitHub website. After running `git status` (next image) I'm afraid I might have comitted both reports in both branches. Is that possible? If positive, can I delete this commit or start over?

[image]

Sorry, sorry... (and thank you very much). 

Julia

Albert-Alvarez, 2021-02-21:
Have you indicated a message during the commit?

```bash
git commit -m "Your message"
```

What is shown in your terminal is the common form to indicate a message when no message is indicated in the commit command.

juliameca, 2021-02-22:
I didn't! I thought maybe the message wasn't necessary here. My mistake. 

Everything is uploaded now. Thank you for your time!

closed 2021-02-22 by Albert-Alvarez