TheAlbertDev/MASB #10: Error when switching branches

esolegal, 2021-02-22:
Hi, Albert!

I have an issue whenever I try to switch branches. After having done the commits and the push, when trying to switch from one branch to another I receive the following message. 
Do you know what might be the issue?

Thank you!
[image]

Albert-Alvarez, 2021-02-22:
Hi Edu!

Your problem is a common problem for Mac users: the [`.DS_Store` files](https://wikipedia.org/wiki/.DS_Store).

`.DS_Store` files are files that Macs generate to set the attributes of a folder, such as its background image or icons. Junk. Therefore, we will ignore these files in our version control system (Git).

To do this, using Notepad on Windows or TextEdit on MacOS (making sure that the latter is in [normal text mode](https://support.apple.com/guide/textedit/txted1063/mac)), we'll create a [`.gitignore` file](https://git-scm.com/docs/gitignore) in the root folder of our repository (in your case it will be the `masb-p01-gg` folder). The `.gitignore` file contains indications of which files should not be tracked by Git (and whose changes we won't save).

In the `.gitignore` we have created, we add `.DS_Store` to ignore them.

When we do a `git status` ⇒ `git add -A` ⇒ `git commit`, the `.gitignore` will be saved and the `.DS_Store` files will be ignored.

Let me know if you've been able to fix it 😉

Albert-Alvarez, 2021-02-22:
Your issue is very common. I change the title of your issue in order to make it easier to find it.

esolegal, 2021-02-22:
Yes! When I created the .gitignore it didn't appear in the folder (I checked it by writing -ls a on the terminal), so I  looked for info([and found this](https://guides.codepath.com/ios/Using-Git-with-Terminal). Up until now it has solved the issue.
Thank you!

closed 2021-02-24 by Albert-Alvarez