TheAlbertDev/MASB #9: Error P1 STM

mariapenon, 2021-02-22:
Hi, I'm getting this error in the beginning of the STM practice 1: 
[image]
If I look at it closely, the error seems to be in the name of the folder in which I save all the practices. 
[image]
Is there any way to fix it? If I change the name of the folders, will it alter my repository?
Thanks.

Albert-Alvarez, 2021-02-22:
The only way to fix it is to place your repository in a path with English alphanumeric characters only. It's easy to solve. Just simply copy/paste your entire folder (`masb-p01-ecologiques`) and place it in another location free of "strange" characters into its path. For example, you can create a folder called `workspace` into your root folder. E.g., if you work on a Windows computer, create a folder on `c:\workspace\` and paste there your folder `masb-p01-ecologiques`. Then, you can work normally from your new location.

Let us know if you have been able to solve your errors 😉

mariapenon, 2021-02-22:
The error has disappeared creating a new folder with english alphanumeric characters, and doing a copy/paste. Thank you!

closed 2021-02-22 by Albert-Alvarez