TheAlbertDev/MASB #13: Failed push

juliameca, 2021-03-04:
Hi Albert, 

I tried to `git push` my final report but it returns the following error: 

[image]

Earlier today, I commited the report by mistake when it was not finished. I deleted it manually from GitHub. It's possible that this is the origin of the error? I tried `git pull` (as the terminal recommends), but after the following message I'm afraid the branches merge... 
[image]

How can I solve it? Thank you

juliameca, 2021-03-06:
I saw in the "history" of my B-branch that the merge did happen. The thing is, that the merge is between `B-julia` and `B-julia`, it didn't merge with the `master`. 

[image]

I tried again to commit the report, but I cannot see it in my history although it was (in theory) done: 

[image]

Is this a serious problem?

Albert-Alvarez, 2021-03-07:
In principle, there is no problem to merge a branch with itself. That seams (form your first post) that you have started to work from a PC that didn't have the local repository updated with the last changes in the remote repository (maybe you edited something from another PC or from the website directly). When you try to push, Git told you that you missed some updates from the remote repository, and you need to incorporate them to your local repository before push. The terminal told you what command to use (`git pull`). When you do so, a merge commit is performed to incorporate the missed updates from the remote repository.

For me, it looks fine 😉

closed 2021-03-20 by github-actions[bot]