TheAlbertDev/MASB #14: No output in the Serie Monitor P04-Arduino

HelenaRD33, 2021-03-06:
Good afternoon Albert,

When executing the code for the serial communication in the Arduino part of the lab 4 I don't get any values as ouputs in the Serie Monitor when turning the potentiometer knob.
I have made sure that the code was as in the README document and I think other groups have the same problem.

Thank you in advance,

Helena

Laiaborrell, 2021-03-06:
I have the same problem, could not fixed it either

Laiaborrell, 2021-03-07:
Hi Helena, I have changed these value on the image to 9600 and now the terminal shows the outputs. 
[image]

HelenaRD33, 2021-03-07:
Thanks Laia!!

closed 2021-03-20 by github-actions[bot]