TheAlbertDev/MASB #20: Error: "storage size of myHandles isn't known"

Laiaborrell, 2021-04-13:
When debugging my code of last session (7) this error keeps appearning. I have already asked some classmates and no one has a clue of what could be happening. Also, I have reviewed that I have not left anything from the code, but obviously I am missing something.

[image]

Any help please?
Thankyou!

Albert-Alvarez, 2021-04-15:
What was the problem/solution?

Laiaborrell, 2021-04-15:
I just did all the session again, but could not find any differences between the code which gave the error and the new one. Now it does not show any errors when debugging. However, it is still not working. Nothing happens when I send the " 0201010101010103D03F010101010103E03F010101010114E0BF027B14AE47E17A843F7B14AE47E17A743F00" code from the Cool Term in "Hex" although I have made sure that the baudrate is set at 115200 and that the breakpoint is placed correctly in te NOP() function. I have no clue of what could be happening.
Laia

Albert-Alvarez, 2021-04-15:
I have regenerated the code from the `.ioc` file and at least the microcontroller has entered in the `HAL_UART_RxCpltCallback` when a message from CoolTerm is sent.

Anyway, the program doesn't work because all the index increments are missing: `rxBufferIndex++`, `txBufferIndex++`, `i++`,.... This is probably because you've copied and pasted the code from the script, you wanted to remove the `+` from the code by doing a find/replace and you have also removed the `+` signs from the increments unintentionally. Fix that and check if it works already. Confirm if that's the case, if not, make a new commit with the index increments fixed and I'll take another look.

Laiaborrell, 2021-04-16:
Thankou, might have forgotten them when doing again the session to fix the first error. I have placed again the increments. It seems like it is working now. 
Laia

Albert-Alvarez, 2021-04-19:
Glad to hear that! I close the issue since it has been resolved 😉

closed 2021-04-19 by Albert-Alvarez