TheAlbertDev/MASB #23: P8 doubts

mariamor99, 2021-04-26:
Hi! I have some questions about i2c. 
According to the i2c protocol, we have to register the information to a slave register address, which can be found in the datasheet of the sensor. I've been looking at the datasheet you shared with us and I've found two register addresses so I don't know which one I should use nor the differences between them. 
This is one:
[image]

And the other one: 
[image]

Moreover, I'm assuming that the output data is raw so it will need to be converted. Also in the sensor datasheet, I've seen this code: 
[image]
This code uses the first register address I've uploaded, nonetheless, I don't quite understand if it's applicable to Arduino. 
Hope my doubts are clear :) 

Thanks

Albert-Alvarez, 2021-04-26:
Indeed, it is just as you say. The registers `0xFA...0xFC`, `0xFD...0xFE` and `0xF7...0xF9` contain the results of temperature, humidity and pressure, respectively. These are read-only registers (they cannot be written to) and these results are directly the value given by the internal ADC of the BME280. You already know from previous practices, that ADCs return a digital integer value and must be converted to the final result using formulas.

In this case, the manufacturer, instead of giving us a general formula, gives us some specific parameters for each chip: `dig_T1`, `dig_T2`,..., etc., which are the ones you have attached in the issue. Each sensor that is manufactured, no matter how good the manufacturing process is, there is always a tolerance between them in their response. The manufacturer, during the manufacture of the sensors, has calibrated the sensors and has added in the `dig_T1`, `dig_T2`, etc. registers the values of their calibration. These values are read only and exclusive for each sensor/chip.

You have to read the sensor registers and the calibration registers and, with them, use the formulas indicated in the datasheet to obtain the final result.

For example, if you get the temperature registers (`0xFA...0xFC`) and the compensation registers for temperature (`0x88...0x8D`), the formula would be:

```C
uint8_t temp_msb; // Register 0xFA
uint8_t temp_lsb;   // Register 0xFB
uint8_t temp_xlsb; // Register 0xFC

int32_t adc_T; // Union of temp_msb, temp_lsb, and temp_xlsb

uint16_t dig_T1; // Union of registers 0x88...0x89
int16_t dig_T2; // Union of registers 0x8A...0x8B
int16_t dig_T3; // Union of registers 0x8C...0x8D

// Here you get the value of the registers using I2C/SPI and save them in the corresponding variable.
// Also, merge the read array of bytes temp_msb, temp_lsb, and temps_xlsb in a single 32-bits
// variables called adc_T (unions?)

double temp;
double var1;
double var2;

var1 = (((double)adc_T)/16384.0 - ((double)dig_T1)/1024.0) * ((double)dig_T2);
var2 = ((((double)adc_T)/131072.0 - ((double)dig_T1)/8192.0) *
    (((double)adc_T)/131072.0 - ((double)dig_T1)/8192.0)) * ((double)dig_T3);

temp = (var1 + var2) / 5120.0;
```

> Please note that I have written this code directly in this issue and have not compiled/tested it. I may have skipped a parenthesis, semicolon or number. The code in the datasheet is the only one that is 100% sure to work.

mariamor99, 2021-04-26:
Perfecte gràcies. Ara ho provaré

Obtener Outlook para iOS<https://aka.ms/o0ukef>

Aquest missatge, i els fitxers adjunts que hi pugui haver, pot contenir informació confidencial o protegida legalment i s’adreça exclusivament a la persona o entitat destinatària. Si no consteu com a destinatari final o no teniu l’encàrrec de rebre’l, no esteu autoritzat a llegir-lo, retenir-lo, modificar-lo, distribuir-lo, copiar-lo ni a revelar-ne el contingut. Si l’heu rebut per error, informeu-ne el remitent i elimineu del sistema tant el missatge com els fitxers adjunts que hi pugui haver.

Este mensaje, y los ficheros adjuntos que pueda incluir, puede contener información confidencial o legalmente protegida y está exclusivamente dirigido a la persona o entidad destinataria. Si usted no consta como destinatario final ni es la persona encargada de recibirlo, no está autorizado a leerlo, retenerlo, modificarlo, distribuirlo o copiarlo, ni a revelar su contenido. Si lo ha recibido por error, informe de ello al remitente y elimine del sistema tanto el mensaje como los ficheros adjuntos que pueda contener.

This email message and any attachments it carries may contain confidential or legally protected material and are intended solely for the individual or organization to whom they are addressed. If you are not the intended recipient of this message or the person responsible for processing it, then you are not authorized to read, save, modify, send, copy or disclose any part of it. If you have received the message by mistake, please inform the sender of this and eliminate the message and any attachments it carries from your account.

mariamor99, 2021-04-26:
Hi! I just tried this and I don't quite understand when dig_T# come into play. 
The thing is I'm using the function Wire.write() to write temp_msb, temp_lsb and temp_xlsb and then I make an array adc_T with these 3. But when I write your code I get a Temperature = 0ºC. I assume it is because I'm not writing anything on the dig_T#. 
So the problem is I haven't understood the difference between 0xFA and 0x88. 
Thanks

Albert-Alvarez, 2021-04-26:
`dig_T#` must be read because you need it in order to calculate the temperature. With this sensor, you only must read the registers. You can't write on them. However, you must use the `Wire.write()` function to send the register addres to be read. Then you have to read the returned value using `Wire.read()`.

For example, to read `dig_T1`:

```C
uint16_t dig_T1;

Wire.beginTransmission(0x77);
Wire.write(0x88);  
Wire.endTransmission();

Wire.requestFrom(0x77, 2);

while(Wire.available() < 2) {};

dig_T1 = Wire.read();
dig_T1 = dig_T1 | (Wire.read() << 8);
```

closed 2021-05-09 by github-actions[bot]