TheAlbertDev/MASB #27: STM prescaler (clock SPI)

Laiaborrell, 2021-04-29:
Hi!
I was starting the `.ioc` file configuration and in SPI there is a `prescaler (Baud Rate)` parameter for the clock signal of the SPI. I could not find information of which value should be given to this parameter. Any help?
Pd: it takes values of 2^(x). 
Thanks,
Laia

Albert-Alvarez, 2021-04-29:
The prescaler serves to divide the clock frequency sourced to the SPI peripheral in order to achieve the desired frequency transmission during the communication.

First, we start finding which is the frequency of the clock signal sourced to the SPI peripheral. From the [datasheet of the STM32F401RE (page 14)](https://www.st.com/resource/en/datasheet/stm32f401re.pdf) we can see that SPI peripherals use the clock APB1 or APB2 (depending on the SPI peripheral used). For example, SPI2 uses APB1.

[image]

Then, we can go to the STM32CubeIDE and check which frequency is configured for the APB2. We can see that is configured with 42 MHz.

[image]

If we want a frequency transmission equal to the example in the other issue (500 kHz), we obtain a prescaler equal to 84. We have 84 available as prescaler? No. We can use 64 (frequency transmission equal to 656.25 kHz) or 128 (frequency transmission equal to 328,125). You can choose either of the two as the obtained frequencies are in the [frequency range compatible with the sensor](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme280-ds002.pdf).

[image]

Laiaborrell, 2021-04-30:
Ok! Thank you

closed 2021-05-03 by Albert-Alvarez