TheAlbertDev/MASB #28: wake up the sensor in STM

Laiaborrell, 2021-05-01:
Hello!
When implementing the code for reading temperature in STM, the T adc value (before compensation formula) that is reading is continuously -256 (11111111111111111111111100000000), as happened with arduino when the sensor was not properly woke up. However, I am calling the following function inside the setup loop:

```
void wakeUp(uint8_t reg, uint8_t value){ 
	uint8_t txBuf[2]={reg & ~0x80, value};
	HAL_GPIO_WritePin(BME_CS_GPIO_Port, BME_CS_Pin, GPIO_PIN_RESET);
	HAL_SPI_Transmit(&hspi1, txBuf,2,10);
	HAL_GPIO_WritePin(BME_CS_GPIO_Port, BME_CS_Pin, GPIO_PIN_SET);
}
```

-----------------------
```
void setup(void) {
	wakeUp(CTRL_HUM,valHUM);
	wakeUp(CTRL_MEAS,valMEAS);
}

```
and still not working... . 
Context: 
CTRL_HUM=0xF2
valHUM=0b00000001
CTRL_MEAS=0xF4
valMEAS=0b00100111

I don't know if I am not using properly the HAL_SPI_Transmit function for writing in the corresponding registers or if it is something else. The full code is commited in the git repository in case it is needed.
Laia

Laiaborrell, 2021-05-01:
Okay I was not selecting the CS pin correctly!

Albert-Alvarez, 2021-05-03:
Glad to hear that you solved it 😉

closed 2021-05-03 by Albert-Alvarez